wulfy23/rpi4 #1: Quetion - upgrade

truefriend-cz, 2020-11-08:
Hi. Your builds is great!
Please how i can upgrade packages/software manually?

One package - (luci-theme-argon) is new version but not update from CLI (opkg update).
https://github.com/jerrykuku/luci-theme-argon/releases

Thanks

wulfy23, 2020-11-08:
hello,

wifi issues are known and documented on the README... i've tested the newer argon theme and it was not enough of a change to go through the trouble of porting the current css mods... but you are welcome in both cases to un-install / install whatever works for you... you might need to ask jerrykuku re: the install steps because... it's not build specific and http ipks do work...

good to see someone running the squashfs image...  a pioneer ;)

truefriend-cz, 2020-11-09:
Hello,

There are no other bugs but this is a crucial feature for me to use your build. Otherwise, all I have to do is use the original one, but there is a lot of work with it and I am not such a skilled worker yet. If you need it, I'll be happy to help with something.

closed 2020-11-10 by truefriend-cz