wulfy23/rpi4 #2: Support ICS Advent Ethernet USB adapter

chinmaythosar, 2020-12-04:
I believe the official kmod for it is kmod-usb-dm981 however the module I have is SR9800 which is set to ‘m’ in the Linux kernel (is it supposed to be set to y?). Is it possible to add support for this in the ‘extra’ build ?

wulfy23, 2020-12-04:
this one can go in all the builds as it's related to firstboot internet/basic functions...

which one exactly? ( will put in next build... just check packages.txt in the folder of you are not sure )

#kmod-usb-net-sr9700
#kmod-usb-net-dm9601-ether

have you tried both of the above? do you know if they conflict?

chinmaythosar, 2020-12-05:
I don't think dm9601 is the correct one, my bad.

When I run lsusb on Raspberry Pi OS it shows :

Bus 001 Device 003: ID 0fe6:9800 ICS Advent 

On Mint on my x86 box it says SR9800 (the code above). I tried both the kmods above with a custom openwrt build but I was not able to detect it (works otherwise in linux) ... I think SR9800 is the successor to SR9700 so I think the more apt module would be that one. However if you look at the the linux kernel it has the driver necessary (https://cateee.net/lkddb/web-lkddb/USB_NET_SR9800.html).

I think its worth a try to include SR9700 and see if the 9800 works but I think 9800 kmod would be needed.

wulfy23, 2020-12-05:
ok, seems this isn't bundled for OpenWrt at the moment...

i've made a patch... but would need someone to test it... i'll ask on the forum...

wulfy23, 2020-12-08:
well... that could take quite some time if someone tests it... seeings as you mention mint... this will get you very close to building a the module within the buildroot... if it fails apologies... but most of the info you'd need to build and test ( probably on a spare sdcard... )...

```bash
#!/bin/bash
#NOTE: save this file as NAME.sh and run 'chmod +x NAME.sh; ./NAME.sh #twice'
reqPKGSdebian10="build-essential subversion libncurses5-dev zlib1g-dev gawk gcc-multilib flex git-core gettext libssl-dev ccache rsync unzip wget curl quilt patch git git-core"
if [ ! -f ./.prereq ]; then
	echo "please install (asroot): $reqPKGSdebian10 curl and re-run(asnormaluser)" && sleep 6 && touch ./.prereq
fi
if [ ! -d "./openwrt" ] && [ ! -f ./BSDmakefile ]; then
	git clone https://git.openwrt.org/openwrt/openwrt.git || (echo "git issue" && exit 0)
	cd ./openwrt
	./scripts/feeds update -a
	./scripts/feeds install -a
	mkdir -p ../dl; ln -s ../dl dl
	touch ./.setup
elif [ -d "openwrt" ]; then
	cd openwrt
fi

if [ ! -f ./.config ]; then

cat <<'PPP' > .config
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_usb-net-sr9800=y
PPP

if [ ! -f ./sr9800v1.patch ]; then
cat <<'VVV' > sr9800v1.patch
diff --git a/package/kernel/linux/modules/usb.mk b/package/kernel/linux/modules/usb.mk
index eaf9eadb28..44d7f412f1 100644
--- a/package/kernel/linux/modules/usb.mk
+++ b/package/kernel/linux/modules/usb.mk
@@ -1335,6 +1335,21 @@ endef
 $(eval $(call KernelPackage,usb-net-sr9700))
 
 
+define KernelPackage/usb-net-sr9800
+  TITLE:=Support for CoreChip SR9800 ethernet devices
+  KCONFIG:=CONFIG_USB_NET_SR9800
+  FILES:=$(LINUX_DIR)/drivers/$(USBNET_DIR)/sr9800.ko
+  AUTOLOAD:=$(call AutoProbe,sr9800)
+  $(call AddDepends/usb-net)
+endef
+
+define KernelPackage/usb-net-sr9800/description
+ Kernel module for CoreChip-sz SR9800 based USB 2.0 10/100 ethernet devices
+endef
+
+$(eval $(call KernelPackage,usb-net-sr9800))
+
+
 define KernelPackage/usb-net-rndis
   TITLE:=Support for RNDIS connections
   KCONFIG:=CONFIG_USB_NET_RNDIS_HOST
VVV
fi
if [ ! -f ./.sr9800v1.patch.applied ]; then
	echo "apply ./sr9800v1.patch" && sleep 2 && git apply ./sr9800v1.patch && touch ./.sr9800v1.patch.applied
fi; if [ ! -f ./.sr9800.patch.applied ]; then echo "oops patch issues" && exit 0; fi

make defconfig

echo "starting build... takes ~1hr if-ok > bin/targets...for img"; sleep 3
make -j12 v=s
```

wulfy23, 2020-12-08:
may as well close this issue also as it's primarily an openwrt / kmod thing now... should the module become available i'll be sure to include it in the build... cheers.

closed 2020-12-08 by wulfy23

chinmaythosar, 2020-12-09:
No problem thank you ...

chinmaythosar, 2020-12-13:
https://github.com/chinmaythosar/OpenWRTBuilds

Here is a build in case anyone has the same adapter and wants to use it with OpenWRT on the Pi4