wulfy23/rpi4 #3: is there any way to have this as a docker, or pihole as a docker with it?

molly78, 2021-01-29:
Seems like a rpi4 can do 100x more than what openwrt requires.

Just wondering if the magic that is pihole can be combined on the same pi is all

wulfy23, 2021-02-01:
thanks for the comment...

were I in need of similar requirement... i'd invert the status quo... and dockerize / lxc / other the openwrt instance on raspiOS/Other... a few options are available now for arm virtualization... but openwrt does not make the best/most user
friendly HostOS...

one user requested rootfs.tar.gz which I did/do provide for this requirement... I believe they use docker on raspbian from memory...

fwiw these things are so modular... I can see a student / cost~power~space concious person making use of this topology... but I'm  bit old school and network centric so lean towards plonking in additional hw-units for each task... just makes fault tolerance / troubleshooting that much easier for core~production devices...

molly78, 2021-02-01:
Thanks for getting back to me

Yes, a single pi4 handling dockers such as openwrt, pihole, home assistant, dokuwiki, Kodi database, would be awesomely handy for me!

Time to do some more research as each of these really doesn’t require much.

wulfy23, 2021-02-01:
no worries...

i'll try to remember to upload the rootfs.tar.gz more often...

i'm just polishing off beta PARTUUID support which might help in some of those applications as well...

closed 2021-02-01 by wulfy23

TalalMash, 2022-04-16:
Hello! Feel free to copy/modify my LXC Pi-hole LuCI configurator.