wulfy23/rpi4 #4: USB Ethernet

unclehowell, 2021-02-06:
You think you could see if you can get USB Ethernet working with OpenWRT on Pi4. 
That would be awesome

wulfy23, 2021-02-06:
it already works... it is quite complicated tho' due to cable specifics and the hardware wiring on the device sadly... to tweaking the build to enable it out of the box would imply advanced knowledge/hardware of/for this by users most of whom wont have...

like all things with the build tho'... propose the necessary code / package changes on the forum thread and if 3 or more people really want / will use this feature i'd be happy to add a few specific configs or setup automation for this :)

otherwise just install any packages that are needed, make sure you have the right cables and configure your settings...

peace

unclehowell, 2021-02-07:
peace

wulfy23, 2021-06-29:
fwiw... i've just enabled this by default ( driver side )
and while its really not worth the hassle due to the
complicated and risky cable complications...

-requires reboot after firstboot

-you still need /etc/config/network i.e.;

config interface 'gether'
	option proto 'static'
	option ipaddr '10.11.12.3'
	option netmask '255.255.255.0'
	option macaddr 'CC:DD:32:77:DD:65'
	option device 'usb0'

will get you setup on any (newer) build

wulfy23, 2021-07-02:
i've just disabled this by default... not a good idea... but there are some more drivers and most of the setup logic ready to go for anyone with the need... (and skills to handle potential fallout)

closed 2021-09-17 by wulfy23