wulfy23/rpi4 #6: USB Boot

Ash-Ketchum1192, 2021-02-18:
Not really an issue but I'm a GitHub noob so I'm sorry! I would like to boot from my USB 3 stick as the R/W speeds are much faster.

I wrote the image to a USB and updated fstab /root to be /dev/sda but unfortunately it gets stuck in a boot loop. Is there a proper way to do this?

TIA
Alex

wulfy23, 2021-02-19:
n0 need to update fstab... openwrt does not really use it...

just gunzip the 'fac' partuuid variant... dd to disk... and make sure there is no mmc in the mmc slot on the pi ( or the mmc has no OS on it ) if you have a default boot order...

the process is more complex for non 'fac' installs...

wulfy23, 2021-02-22:
Assuming this is resolved

Ash-Ketchum1192, 2021-02-22:
Thabks for your reply! Unfortunately I tried extracting the fac image,
wrote it to the USB and it boots but gets stuck during the boot sequence.

I tried burning Raspian to the USB just to check and it boots completely
fine so it can't be the USB.

Any ideas?

AnthonyMann, 2021-02-22:
Also check you've upgraded to a recent firmware, USB boot was added only a few months ago.

You'll need a Raspbian sd image booted, then
`sudo rpi-update`

Ash-Ketchum1192, 2021-02-22:
Thanks, I did enable it a while back and as I said after Openwrt failed to
boot I tried Raspian on the USB and it booted fine.

AnthonyMann, 2021-02-22:
> Thanks, I did enable it a while back and as I said after Openwrt failed to
> boot I tried Raspian on the USB and it booted fine.
> 
> 
> > Also check you've upgraded to a recent firmware, USB boot was added only a
> > few months ago.
> >
> > You'll need a Raspbian sd image booted, then
> > sudo rpi-update
> >
> > <https://github.com/wulfy23/rpi4/issues/6#issuecomment-783168733>, or
> > unsubscribe
My bad, I deleted my comment as after Re-reading your post it was clear you were already booting raspian that way.

wulfy23, 2021-03-28:
"gets stuck during the boot sequence"

apologies for the delay... just noticed this now > reopen issue ( thanks anthony )

ideally... you should have a console connection... but essentially (described a little / lot in bottom of the README@builds )... if you can put the usb stick in a PC... and list the PARTUUID ( on linux 'blkid' will do this ) ... you then need to mount the first partition and show me the content of 'cmdline.txt'....

another useful addition... is the boot led now operates... and it goes through 4 sets of patterns as it boots ( firstboots )... **so if you noticed the patterns changing?**... then we can find out a little from that...

-fast blink
-steady regular blink (stops for a little)
-pulses / throbs

but... after all those notes... my gut's telling me it's **something to do with the download / writing process**... **can you describe exactly how you wrote the image?** also... if you have a usb-mmc-reader... these are really good because you can alternate between using that and plugging the (mmc)disk into the mmc slot...

wulfy23, 2021-04-01:
probably too old... anyway additional notes maybe useful for someone else...

please seek future support re: this important issue on the forum thread....

closing...

closed 2021-04-01 by wulfy23

Jarvis-1487, 2021-05-25:
Sorry, after moving house I finally managed to test this! I forgot that OpenWRT defaulted to 192.168.1.1 and therefore I couldn't ping/access it. Once I put myself on the same subnet, it worked instantly and was probably always working! Sorry if I wasted your time!