wulfy23/rpi4 #13: /etc/custom/firstboot/35-masqaddrouter breaks dnsmasq, when ip of interface with cidr postfix

BenjaminBeichler, 2021-08-30:
Since the IP of an interface can be given in cidr notation this script breaks the dnsmasq config. 

Line 19 of that script get the interface ip, which can contain cidr postfix.

`BRLANIP=$(uci -q get network.lan.ipaddr)`

I'm not totally aware, why this whole script is needed anyway (dnsmasq should resolve the routers hostname anyway) but for a first quickfix this line should do the trick.

`BRLANIP=$(uci -q get network.lan.ipaddr| cut -d \/ -f 1)`

I did not found directly the script in the Repo, therefore here only an issue and no pull request.

wulfy23, 2021-08-30:
Thanks for the info mate. Will checkout and attempt to resolve or optionalize if possible...

It's a generic hostname to aid in support and for simple users...

https://forum.openwrt.org/t/rpi4-community-build/69998/837?u=wulfy23

wulfy23, 2021-08-31:
note: if you are patching this locally until the next build there is an identical script /etc/custom/startup/35-masqaddrouter ... this one technically should run from hotplug to catch runtime lanip changes... but the initial hack was to run it on a startup+firstboot

BenjaminBeichler, 2021-08-31:
Yeah, actually my fix was too remove the /24 from my config, since it was redundant. It only bugged me, that after every update my config broke and I searched the offending script (s) 😉 I found both, but I assumed it may be this one, because the problem was only after update. But if course the bug is in both.

closed 2021-09-01 by wulfy23