wulfy23/rpi4 #16: Thank you, thank you, thank you

molly78, 2022-03-10:
Thank you for the update with the critical patches wulfy23. That was extremely kind of you :-)

zer0fl0w, 2022-03-10:
`issue` my thanks!
_...for keeping it alive and healthy_

wulfy23, 2022-03-12:
i'm not sure thanks are justified in this case... i'd be pretty pissed being orphaned with a custom build with known CVE's (and fixes available) for 3-5 days...

in any event you are all most welcome ( i run this build too you know ;) )

and thanks for your patience...

good news is around 25 (somewhere near 50%) of build users updated within 24 hours...
and around 15 more the next day... (close to 70%?)
(the rest would likely be on 21.02 or migrated away already)

fantastic result...

[image]

molly78, 2022-03-12:
Thanks are very much in order :-)

having a router that works as well as it does, no stupidity to it, no locked features…. It’s a bit of happiness

ds2288, 2022-03-17:
Thanks for all your work. Sorry to hear about the rubbish that happened over at the forums. If you ever built another software package for the rpi4 based on another project (OPNSense, etc) I’d use it for sure, rather than support a brand who’s developers don’t appreciate what senior members of the community offer.

BIGFAAT, 2022-03-23:
Thank you very much for continuing the build for the moment. Github should be more than enough to provide support even for beginners.

By the way: fuck bullies.

grischard, 2022-03-28:
I see Taiwan getting coded as `partly` in that screenshot and smile, knowing exactly what happened. Thank you for this update.

zer0fl0w, 2022-04-03:
thank you for 5.7.31-2, smooth sailing!

Entropy512, 2022-04-22:
Wow.  I just checked the thread and saw what happened.  I never ran your build myself (vs. a custom self-compiled build tailored to some internal corporate needs), but your build thread was always informative as to various Raspi-specific quirks (such as the endless country code fiasco).

I am wondering if the person who DDOSed you was the same anonymous coward who privately NAKed the Pi Zero 2 W builds but refused to go on record, and said something along the lines of "i do not touch brcm shit".

zer0fl0w, 2022-04-27:
thank you for 5.7.37-2, smooth sailing!

zer0fl0w, 2022-05-08:
thank you for 5.7.50-3, smooth sailing!

netmask69, 2022-05-24:
I am happy to use this distribution for a few months, thanks for your work and thanks for the new version 7.1.13-3

zer0fl0w, 2022-05-26:
also happy to report that 7.1.13-3 is running fine in my environment. thanks a lot for your continuous support.

zer0fl0w, 2022-07-05:
thank you for 7.1.31-3, smooth sailing!

netmask69, 2022-09-27:
thank you for 7.1.37-7

thanks a lot for your continuous support.

netmask69, 2022-10-19:
thank you for 7.1.59-2

thanks a lot for your continuous support.

netmask69, 2023-01-16:
thank you for 7.1.75-11
no issues for me, qosify ok, mwan3 ok, i not use dnmasq with ipset

thanks a lot for your continuous support.

netmask69, 2023-06-26:
Thank you for 7.3.95-5 kernel 6.1
No issues for normal use and for sqm, just a problem with adgurdhome after reboot, i tried putting s99 and k120 but it doesn't solve the problem. Fixed with a crontab script.

thanks a lot for your continuous support.

wulfy23, 2024-01-03:
will close... and **thankyou all** for being such patient, open, understanding people

closed 2024-01-03 by wulfy23