wulfy23/rpi4 #17: rpi-perftweaks.sh syntax error

Synapse07, 2022-03-22:
Hi, thank you for continuing the builds.

I've found some things. firstly the perftweaks scripts output mentions some syntax errors,

> [root@RPi4B /]# rpi-perftweaks.sh
uci: Entry not found
INTERACTIVE rpi-perftweaks.sh> GOVERNOR[ondemand]: quick  upthresh:21 downfac:6 minfreq:1000000 maxfreq:2000000
INTERACTIVE rpi-perftweaks.sh> TASKSET:  collectd:26595>3 uhttpd:13614>2 odhcpd:21035>3 dropbear:24279>3 dropbear:20035>3 dnsmasq:26332>3 dnsmasq:26324>3
INTERACTIVE rpi-perftweaks.sh> MMCINTs:  c:c>32
INTERACTIVE rpi-perftweaks.sh> ETH0INT:  39:1=1 40:2=2
INTERACTIVE rpi-perftweaks.sh> LANIFACE:eth0 WANIFACE:eth1
INTERACTIVE rpi-perftweaks.sh> sysval: eth0/queues/tx-0/xps_cpus[1=1]
INTERACTIVE rpi-perftweaks.sh> sysval: eth0/queues/tx-1/xps_cpus[2=2]
INTERACTIVE rpi-perftweaks.sh> sysval: eth0/queues/tx-2/xps_cpus[4=4]
INTERACTIVE rpi-perftweaks.sh> sysval: eth0/queues/tx-3/xps_cpus[4=4]
INTERACTIVE rpi-perftweaks.sh> sysval: eth0/queues/tx-4/xps_cpus[2=2]
INTERACTIVE rpi-perftweaks.sh> sysval: eth0/queues/rx-0/rps_cpus[7=7]
INTERACTIVE rpi-perftweaks.sh> sysval: eth1/queues/rx-0/rps_cpus[7=7]
/bin/rpi-perftweaks.sh: line 632: syntax error: unexpected "fi" (expecting "then")

also, when logging in this happens,

> [root@dca632 / 53°] exit
exit
root@RPi4B /51#

i don't think this was like is in previous builds, ssh'ing would directly go to root@RPI4B.

wulfy23, 2022-03-22:
first two fixed(-ish) **thankyou!**
(fyi they dont really effect much - some sloppy coding on my part... looks like you uninstalled sqm?)

last one i think you typed exit right? and you've re-installed from factory?

```
grep '^RCSHBASH' /root/wrt.ini
```

if all are true you can comment out the line above... and test re-logging in...

Synapse07, 2022-03-22:
> if all are true you can comment out the line above... and test re-logging in...

That worked. 

> looks like you uninstalled sqm?

yep. I don't have any use for it.

wulfy23, 2022-03-22:
turns out you may not have re-installed from factory...

my terse notes in CHANGES mention 'backport' (means i'm retrospectively turning on or at least was attempting to... I will double check over the logic for that 'backport' and probably cease doing so...)

thanks again...

closed 2022-03-26 by wulfy23