wulfy23/rpi4 #18: request add kmod-inet-diag

malikshi, 2022-03-26:
check the details commit on https://github.com/openwrt/openwrt/commit/efc8aff62cb244583a14c30f8d099103b75ced1d thanks you sir

wulfy23, 2022-03-27:
done (newer master builds only)

AFAIK, could potentially be something post-installed? but;

1) i trust your judgement
2) it's tiny

thankyou!

malikshi, 2022-03-27:
> done (newer master builds only)

 Well I already use master build yet updated to latest one.

Thanks you for added the packages, it's related to tools I use and it can be useful for filtering.

Additional for v2rayA they have opkg feeds, is it possible to add the feeds on default?
https://github.com/v2rayA/v2raya-openwrt

wulfy23, 2022-03-28:
as it's related to package restore? and they seem to have support for multiple releases... I will add a new INI option to add those automatically... and we will see how that goes...

```
OPKGFEEDS_EXTRA_V2RAYA=1
```

malikshi, 2022-03-30:
late update

succesfully updated to latest build.

> logic resides and can/could be manually tested/run as below;
> 
> ```
> OPKGFEEDS_EXTRA_V2RAYA=1 /etc/custom/firstboot/53-customrepo-v2raya
> ```

and i tried add `/etc/custom/firstboot/53-customrepo-v2raya` with this code
```sh
wget https://osdn.net/projects/v2raya/storage/openwrt/v2raya.pub  -O /etc/opkg/keys/94cc2a834fb0aa03
echo "src/gz v2raya https://osdn.net/projects/v2raya/storage/openwrt/$(. /etc/openwrt_release && echo "$DISTRIB_ARCH")" | tee -a "/etc/opkg/customfeeds.conf"
```
everytime i open terminal via putty or other tools it's looping 6x times...

wulfy23, 2022-03-31:
that build does not have the changes yet...

all you really need is

```
OPKGFEEDS_EXTRA_V2RAYA=1
```
in /root/wrt.ini

but I will upload the sample in case you want to test it...

malikshi, 2022-04-01:
LGTM
thank you very much sir!