wulfy23/rpi4 #19: irqbalance issue with 5.0.19-157?

avhm, 2022-04-07:
The last release for me was perfect, but the current release seems to have an issue with balancing load across cores.

With the current release running IRQ balance and `PERFTWEAKS="default"` in my wrt.ini, I see almost all load hitting core 0 (it's the same with packet steering enabled or disabled)

Restarting IRQ balance doesn't appear to change this (I would occasionally have to do this before).

My WAN is 600/36, and I run SQM w/ ctinfo_4layercake_rpi4.qos, irqbalance.

If I manually run the `rpi-perftweaks.sh_userfile_simple` script I am seeing the load distributed evenly again, but the performance feels less snappy than the prior release.

Any suggestions? I may have misunderstood the ideal configuration for this scenario!

Thanks once again for a great build, and continuing to keep this updated :)

wulfy23, 2022-04-09:
While there could be room for re-assessing interrupts in general...

...gut feeling says this is exactly as the issue title states;

that there is actually an issue with the irqbalance program itself
(i.e. it's just not working?) in this case your best options are probably revert to prev. build / 21.02 or see if PERFTWEAKS_Gbs=1 improves your performance for now?

... and i/we'll keep an eye out for irqbalance operations until better / report this in RELEASE_NOTES

wulfy23, 2022-04-09:
... and

assuming the above is correct, that still leaves the 'less snappy' stuff on the table (i.e. something else is causing that > after you ran userfile_simple)

which odds on are something elsewhere in the stack... obvious cantidates are qos/network stack type stuff... so we will keep an eye on this probably for a few builds or so...

i.e. keep reverting to the better build and if you have time / energy test some newer ones if/when they come online and report back if something is snappier... (would be great but sooner or later master might become incompatible / problematic in other respects so investing lots of energy in odd stuff might not be a good use of time ;) )

avhm, 2022-04-22:
Sorry for the slow reply, I foolishly sent this just before I went on holiday.

Really appreciate the detailed investigation (as always!). I swear I learn more from you dumping out your diagnosis process than years of reading forum posts and documentation.

Let me have a dig through your diagnosis and see if I can find anything fruitful, I'll try reverting too. 

From a conceptual standpoint I am ambivalent to using either IRQBalance or manually pinning to cores. My only reason for running IRQBalance was that I didn't have the time to experiment with pinning, so it made more sense to let IRQBalance handle it (and it seemed to work well) - I would be more than happy to use a well pinned setup as an alternative!

I'll circle back. Many thanks again.

avhm, 2022-04-27:
Well, through no effort of my own It looks like the latest release has resolved the issue for me!

Now seeing load evenly spread across all cores again (~30% per core @ 560mbps w/ irqbalance)

Just in time for my gigabit line to be installed next week 😆

I did not get any further than the same diagnoses you ran, to no enlightenment.

Were there any fixes made, or was it as simple as a newer irqbalance release?

Once again,

Thank you.

wulfy23, 2022-05-07:
cheers for the update...

no major fixes on my end and I've not looked into the any updates to the underlying packages ( irqbalance, steering logic... maybe pi core code but unlikely... )

I suspect that it was a quirk with irqbalance initialization or similar timing/logic/interaction problem with the tweaks script...

aka (what I ask myself)

-did irqbalance itself decide on some poor runtime cpu affinity setting? (likely) was it timing related? or just a flip a coin type thing... if the former can it be tweaked either via restarting, options or startup delay... (afaik not so much except for the startup delay but that's not really going to help i think)

given how little cpu affinity shifting that needs to (and can) happen on the pi... it's better not to use irqbalance if it continues to do that (often) and gets no newer code...

basically both eth0 affinity to core0 for best latency under 600MB-ish
and
one-eth0 to core0, one-eth0 core1 for over 600Mb-ish

and that's 90% of the job done...

(assuming that is really the issue)

wulfy23, 2024-01-07:
closing, reopen at will

closed 2024-01-07 by wulfy23