wulfy23/rpi4 #20: Please share your story

dangowrt, 2022-04-24:
You write on the project wiki:
> after being DoS'd by a senior openwrt developer i'm no longer active on the forums... and builds are on life support

I'd really like to know more about what happened. I know you may not trust me, but I also know that I'm not the only one being left behind quite puzzled with your sudden leave.
Obviously whatever has happened is really bad because as a community we strongly depend on people like you and the project should be sending you cakes/free beers/flowers on a regular base rather than anyone associated with the project doing anything bad to you.
At least me and some other devs I know in person don't have any clue, and we wonder what we can be done about this situation, also to avoid that this kind of thing won't happen again in future.

Feel free to contact me privately by email, of course. If you feel that all OpenWrt project members including myself cannot be trusted, please be kind and name someone I could approach to at least learn what happened from them.

Thank you!

wulfy23, 2022-04-25:
Thanks for making effort to engage / address et. al.

I guess people are people, and the project as a whole is more
approachable, welcoming, open, flat, forgiving and pragmatic
than most projects / organisations i've engaged with.

All involved have my respect, even those that I may have
taken 'verbose' technical opinions / views with around
the traps...

I respect the decentralized style and open discussion /
disagreement are part of what keeps the project
healthy...

I guess rather than dwell in a single action from a single
person ( they know who they are and are welcome to step up
and explain their actions ), i'd just encourage everyone
to try to keep issues out in the open in a place where
they can be discussed.

My ISP has a 3rd DoS and you are out policy (yes if
you are the victim), so I have 1 remaining chance
before I get terminated, not something I can really
afford to take chances with right now...

Anyways, it's been great being able to give back to
the Linux community via OpenWrt (supporting).

dangowrt, 2022-04-25:
I kinda still don't know who to blame and what the evidence could be. As you are referring to "a senior OpenWrt dev" I wonder what your definition of that term could be. Someone who has been a member of the core group for a long time, ie. even before the 2016 LEDE fork and re-unification? Someone with access to datacentre infrastructure (not many people, but of course includes our ISP as well)?

All we can do with what you gave us is to now start randomly accusing each others without anything concrete in our hands, and I don't think that would help preventing things like that repeating because.

To be more clear: If what you are saying is that someone hostile (within or outside of the core group is yet another question) got access to our infrastructure (and hence would be able to extract your home IP from forum server logs after you login, for example), it would be something really important to know and act upon.

grischard, 2022-04-25:
> All we can do with what you gave us is to now start randomly accusing each other

No, you can also respect how much or how little wulfy23 wants to continue down that path, and not start any witch hunts. If wulfy wants to let it be, let it be.

dave14305, 2022-04-25:
Perhaps @wulfy23 should remove his DDoS claim in README.md if he’s not willing to substantiate it to the OpenWrt team. It creates hostility between his users and the OpenWrt maintainers, and no one knows what to do about it.

dangowrt, 2022-04-25:
Of course, I do understand and respect that @wulfy23 doesn't want to get involved.
However, it's harsh to see an accusation affecting the project as a whole and maybe even hinting to a potential security problem and then being left entirely in the blue regarding any details. It would be in the best interest of the project and also all other users of the forum to not see something like that happening again. And without more details it is impossible to even asses what happened. OpenWrt developers are not a homogeneous group, so even if it was one of "us" that doesn't mean that anyone else in the group would know about it.

TalalMash, 2022-04-26:
No community is perfect, I think this one has some immaturity, I got bashed on OpenWRT IRC server by a lead developer and two old timers for using Windows by sharing a screenshot showing the title bar. OpenWRT has gained popularity recently so I'm not surprised.

dangowrt, 2022-04-29:
Honestly: I also don't like dealing with broadcom systems. Too many proprietary components, starting right at the bootloader, so (similar to Qualcomm ARM SoCs) you feel like dealing with some toy sandbox environment rather than with "real hardware" (you need to enter licence keys to unlock e.g. video encoding/decoding features). So not everybody likes working with that hardware for understandable reasons, and I can also see how Broadcom wasn't exactly friendly or open in the past (and still isn't, strictly speaking).
In the end we are all volunteers, nobody tells anyone what to do and especially not which hardware platform to support or not to support. You just may have a hard time finding reviewers because not enough people care about every platform (that's probably what that big unknown person meant when he wrote "i don't touch brcm shit" -- for me this translates into "do what you like as long as you don't break anything, but I won't lift a finger to help you")

So not everybody may be up to do reviews and actively work to improve a given platform (remember: we are all volunteers, mostly doing this because it's what we like to do), but as long as someone else is up to the review, the code is clean and **licensing unambiguously permits at least unconditional redistribution**, all is fine. So if all what is missing for the PiZero to work better is a **redistributable** firmware blob, please post a patch, I volunteer to review and merge it right away.

Nobody in our group has the power to NAK a submission, especially not off-the-record. I've asked stintel what was the story behind this and will probably know more in a couple of hours. I *guess* that he has access to a more advanced version of the firmware which is **not published under a license permitting unconditional redistribution**.
To understand this better, I thought the best is to just ask: https://github.com/RPi-Distro/firmware-nonfree/commit/feeeda21e930c2e182484e8e1269b61cca2a8451#commitcomment-72496015

Update: this has already been escalated to SFC, see https://github.com/raspberrypi/linux/issues/4723#issuecomment-1101745060

Reading https://github.com/RPi-Distro/firmware-nonfree/commit/8b3067d405a80ee545ad88e91a1f26e65d234264 also makes it clear that this minimal firmware may not be suitable for a number of applications and also **not be aligned with legal requirements to use DFS channels**. Hence I would suggest that (given that redistribution is permitted at all) we can package it in a way that the user may select to use either the full-featured blob which may not be able to handle a lot of clients OR use the minimal blob which lacks lots of features but can handle more clients instead.

Entropy512, 2022-04-29:
"not be aligned with legal requirements to use DFS channels" - Regarding this, you're not understanding the featureset correctly.

Omitting dfsradar does not make the firmware start allowing channels without a DFS check.  It causes those channels to become unusable if CRDA flags them as requiring DFS support.

Entropy512, 2022-04-29:
@blogic fair enough - in the future, could you please include information like that in the commit message so it's easier to track down the origin of the patch if there are questions regarding it?  "fix iwinfo behavior" is a pretty uninformative commit message...

wulfy23, 2024-01-03:
water under the bridge, closing

closed 2024-01-03 by wulfy23