wulfy23/rpi4 #21: Cannot Flash 7.1.31-3

JAAlperin, 2022-06-27:
Red "Flash" link does not complete. Manual "Flash Image" results in "Image Check Failed."  Please see screen shots.

[image]

[image]

[image]

malikshi, 2022-06-27:
Try upgrade through terminal
I had problem with curl yesterday so i flashed same version to fix it.
`sysupgrade -F /path-to-file`
Recommended to save firmware file in /tmp

anipad, 2022-06-27:
Does manual `sysupgrade -F` keep settings and configuration in this build without `-f`?

wulfy23, 2022-06-27:
Thanks for the screenshots and help Malikshi something seems to have changed in the image validation code so until I can take a look as above, sysupgrade -F /tmp/sys.img.gz will flash... adding -R will get you the built in package restore features...

The build pretty much always keeps settings unless you issue -n but this is largely untested.

malikshi, 2022-06-27:
Thanks you for further explanation

wulfy23, 2022-06-28:
Seems certificate i'd been using expired. We were due for a rotation anyway... thanks again for making me aware of the issue...

I'll resolve this weekend...

JAAlperin, 2022-06-28:
Thank you for your generous help.

Regards,

Jeff Alperin
H: 561-666-7789
C: 561-797-9876

zer0fl0w, 2022-07-06:
you are still on 'stable' channel...

put it to:
UPGRADEsFLAVOUR="current"

netmask69, 2022-07-07:
thanks for your suggestion, solved.

wulfy23, 2022-10-12:
seems sorted... re-reading the thread it's possible there is a hidden bug (or I put a 'release' image in stable for a while and that messed it up) that shifted your FLAVOUR to stable...

for the record rather than dig too much now. my code should not change your FLAVOUR... except in the case of firstboot, where if you start with a 'release/stable' image, then 'current' is changed to stable afair... all upgrade after this should leave your setting untouched unless you changed it manually.

you will however as you've noticed, be notified if your flavour is 'current' and you install a stable build... i.e. 'you are running 2.x.x but there is newer 7.x.x' its simply the update system checking where it was told to look.

thanks for the help also... closing re-open if needed.

closed 2022-10-12 by wulfy23

JAAlperin, 2022-10-12:
I’m set for now.  Thanks for everything you do!

Regards,

Jeff Alperin
H: 561-666-7789
C: 561-797-9876

JAAlperin, 2022-10-14:
Sorry.  I've lost track of this.  New version now available, but flash link is gone.  How is this accomplished now?