wulfy23/rpi4 #22: Is there a future for the wulfy23 builds?

jeff47, 2022-07-01:
I am concerned about relying on these builds based on what's been written here.  Is there a future for them, as we move to 22.03?  What is the transition path for us if they are abandoned?

Really appreciate what you've done @wulfy23, but it seems like you've gotten burned and seem discouraged by the whole affair.

hsmptg, 2022-08-04:
I just now realised that this project seems to be somehow "frozen"! :(
I am very sad with that since it is/was a GREAT firmware for my RPi4 router.
The facts are not great since the last issue is from 1/jul and that last code commit is from 25/jun.
Nevertheless, many thanks to wulfy23 and hope he can explain what "... builds are on life support" really means!

jeff47, 2022-08-04:
The Automated System Upgrade Tool will still support future versions, but not quite as neatly packaged as wulfy's.

wulfy23, 2022-10-05:
Valid question. If anyone has concerns about relying in these images due to possible disappearance, code rot or other, the transition path is relatively clear afaik, and that would be to switch to mainline OpenWrt, roll your own via the three most common ways of doing so.

Config migration has been covered in the past several times, significant notice and benefit <> time lost with any migration woes imho has been a debt paid sometime ago to all long and medium term users of this build.

Regarding outlook, there has been no significant change since the re-opening of this repo, and notes therein committing to making key images available while they can be generated without gremlins or other significant negative impact to what users of the build have come to expect.

The key focus of these images has been around: 'fullness (capability out of the box) with a light footprint, ease of upgrade, simplifying setup of common stuff without too much / any impact on peeps who want to do other things..., running master without the hassle of not knowing if it will work or forgetting to install a package on the day you installed it."

The history of releases speaks for itself with relation to stagnation / availability. I think you'd be hard pressed finding anything comparable on the balance of catching faults before they arrive on your system and frequent, if not too frequent update availability. They are not perfect and we've had our share of glitches with various 'features'...

Life support means that they can disappear at any time and break at any time, but 'stable' is likely to see usefulness for sometime yet providing there is a means to distribute it.

On a general note:

-Userbase is approx 10-20 people now, well down from 30-50 of 7 months ago
-Our reliance on iptables... at least in master is perhaps nearing its usefullness / practicability duration
-Webhosting expired a few months ago so we rely on the goodwill of Github for image distribution. Frequent builds stretches this friendship.
-The disk, one click upgrade and package restore capability continue to bring significant benefit to all build users, many other optimisations and tweaks have not seen work in sometime, so their breakage / repair / removal etc. etc. is reliant on the community as a whole ( all 10 of us ;) )...

I hope that helps to answer the questions raised.

wulfy23, 2022-12-01:
closing this as it's answered and rather broad. feel free to open another issue if you have more specific operational or technical queries

closed 2022-12-01 by wulfy23