wulfy23/rpi4 #24: https-dns-proxy

neilo1, 2022-10-11:
Hi,
Thank you for the builds. 
I created this account to ask for your help and tell you that with latest build 22.03.01 https-dns-proxy is not working.

wulfy23, 2022-10-11:
Try removing procd-seccomp and procd-ujail if installed, if still not working use official source / image on an alternate sdcard to test and let me know if works there...

I will remove all https-dns-proxy previous workarounds in future builds to ensure that something stale is not causing the issue.

neilo1, 2022-10-12:
Yes, I removed procd-seccomp and procd-ujail, still not working.

wulfy23, 2022-10-12:
looks like there are others having various issues with this package ( https://www.reddit.com/r/openwrt/comments/y1iyd7/help_httpsdnsproxy_isnt_working_anymore/ ) etc. etc.

only thing you can try which i'd do for any broken / recently operated on package is sliding your config and force reinstalling... something like;

```
mv /etc/config/https-dns-proxy /etc/config/https-dns-proxy_old
opkg install --force-reinstall https-dns-proxy
/etc/init.d/https-dns-proxy restart
```

or something like that... otherwise you might need to shift to stubby or await package fixes... ( same force re-install should pull them in in future )...

wulfy23, 2022-10-15:
sure you can install adguardhome

no idea how it interacts with simple-adblock but if it's configured as a forwarder for your primary dnsmasq instance you should get both however the benefits/need for that are likely minimal?

issue is with upgrades, in that (at least of a year ago, things might have changed a little) every time you upgrade unless your dnsmasq instance has some level of redundancy or the router itself does or adguard is bundled into the image, then you will have to mess around each upgrade (it's not really suited to frequent upgrades unless you heaviliy customize / allow for it)

wulfy23, 2022-10-15:
feel free to pass this onto the package maintainer;

```
----------------------------
WARNING: Running as root without dropping privileges is NOT recommended.
----------------------------
[W] 1665824889.155379 main.c:236 HTTP/3 is not supported by current libcurl
[W] 1665825129.454036 https_client.c:351 66B8: curl request failed with 0: No error
[W] 1665825129.454089 https_client.c:353 66B8: curl error message: Error reading ca cert file p=�� - mbedTLS: (-0x3E00) PK - Read/write of file failed
[W] 1665825129.454127 https_client.c:380 66B8: No response (probably connection has been closed or timed out)
```

also stopping the service left junk in my dhcp config...  definitely broken

neilo1, 2022-10-15:
I Installed and config it and it was working but it was unable to start after reboot, I added in the 
`ENABLEDSERVICES="banip simple-adblock sqm adguardhome" `
but I had to manually start it. So, I removed it.

neilo1, 2022-10-15:
I tried stubby today, but it was also not starting after reboot, I have to manually restart it.  I again messed up my settings and router was refusing to even reboot and remove stubby.

neilo1, 2022-10-16:
upgraded to latest build `release uptodate: 2.1.9-5  twicedaily[refresh]  [backup]  [tty]`

but still https-dns-proxy not working for me. Maybe something is wrong with my configurations.  
and one more thing 
there was used to be editor next to build number on luci page.
`release uptodate: 2.1.6-7  twicedaily[refresh]  [backup] [editor] [tty]` its gone now?

neilo1, 2022-10-16:
And one more update I finally fixed the issue of IPTV stutter and stream stopping after a few mins a month ago. It was an issue with VDSL PTM interface in the modem settings. I was creating separate interfaces for IPTV and internet with vlans. You just need 1 PTM vdsl bridge interface without mentioning any vlan ID. You tried to help me alot and fix that issue for me. 

Created separate vlans in the openwrt interfaces for IPTV and WAN pppoe, Its working fine now.

wulfy23, 2022-10-16:
https-dns-proxy issue has nothing to do with your config

editor is removed as needed to be ported to php8 cause php7 is gone... I was downgrading to that for a while... if you put PHP7=1 in your wrt.ini it may come back next upgrade if it still works...

glad to hear about the IPTV being fixed... now you've schooled me ;)

neilo1, 2022-10-16:
I posted the logs in openwrt forum and heres the reply I got.

`I have sent PRs for main and 22.03 branches with the fix for the garbage in the ca cert file name, given the logs you have provided it should solve the problem: https://github.com/openwrt/packages/pull/19611 https://github.com/openwrt/packages/pull/19612. I'll leave them for others to review/provide feedback for a day or so before merging.

The repo owner for your custom image can also manually build the new package from the updated sources.`

wulfy23, 2022-10-16:
here is the prev working binary if you need it;

https://github.com/wulfy23/rpi4/tree/master/utilities/https-dns-prev

should only need https-dns-proxy with --force-downgrade if  you didn't uninstall broken one first. 

```
wget https://raw.githubusercontent.com/wulfy23/rpi4/master/utilities/https-dns-prev/https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk
opkg install --force-downgradehttps-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk
```

neilo1, 2022-10-16:
`opkg install --force-downgrade ./https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk
Collected errors:
 * wfopen: ./https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk: No such file or directory.
 * pkg_init_from_file: Failed to extract control file from ./https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk.
[root@dca632 / 55°]# opkg install --force-downgrade https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk
Collected errors:
 * wfopen: https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk: No such file or directory.
 * pkg_init_from_file: Failed to extract control file from https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk.`

wulfy23, 2022-10-16:
hmmm... looks like the first wget command did not download anything?

neilo1, 2022-10-16:
> hmmm... looks like the first wget command did not download anything?

I only ran this one `opkg install --force-downgrade ./https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk`

neilo1, 2022-10-16:
`wget https://raw.githubusercontent.com/wulfy23/rpi4/master/utilities/https-dns-prev/https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk
opkg install --force-downgrade ./https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk
--2022-10-16 20:36:32--  https://raw.githubusercontent.com/wulfy23/rpi4/master/utilities/https-dns-prev/https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk
Resolving raw.githubusercontent.com... 185.199.109.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 20708 (20K) [application/octet-stream]
Saving to: 'https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk'

https-dns-proxy_2021-11-22-3 100%[==============================================>]  20.22K  --.-KB/s    in 0.02s

2022-10-16 20:36:33 (1.13 MB/s) - 'https-dns-proxy_2021-11-22-3_aarch64_cortex-a72.ipk' saved [20708/20708]

Downgrading https-dns-proxy on root from 2022-08-12-1 to 2021-11-22-3...
Stopping https-dns-proxy 2022-08-12-1 ✓
Command failed: Not found
Configuring https-dns-proxy.
Starting https-dns-proxy 2021-11-22-3 ✓✓
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/https-dns-proxy is different from the conffile in the new package. The new conffile will be placed at /etc/config/https-dns-proxy-opkg.`

wulfy23, 2024-01-07:
closing, open new one if issues with this package continue... ( on 23.0x.x / master )

thanks again Neilo1

closed 2024-01-07 by wulfy23