wulfy23/rpi4 #25: Want a new host?

molly78, 2022-10-17:
Being able to flash from one click was so useful

what do you need to make it work again? What kind of host you want?

wulfy23, 2022-10-17:
Schucks, that's a mighty kind offer... But if you can't see the flash link and there is an update then that's a recent bug (since 22.02.1-ish aka 1month ~ when https cookie was introduced) I thought i'd fixed.

I will fix it this weekend. (aka do my best to fix... can't guarantee it will work)

Regarding hosting... github is reliable and constraints are just adequate for current user numbers / image counts etc.
(there were one or two tarred up 'features' on the old server that almost nobody used? that are too large for github)

If we end up needing one again, i'm happy to foot the bill.

wulfy23, 2022-10-22:
Just as an fyi for anyone who is missing the graphical flash button (as I have a feeling this is going to take longer than a weekend) ... ssh/command line equivalents for the above functions are;

```
updatecheck.sh
```
and if an update is available
```
rpi-sysup-online.sh -S
```
should do the same thing... ( i'll need to polish / checkup on the state of a 'backup' cmdline tool )

molly78, 2022-10-22:
> 

Thanks - worked perfectly

wulfy23, 2023-01-23:
Some news on this front, as discussed elsewhere, i've bodged up some luci > system > command links for those features that although clunky should accomplish tasks while something cleaner is sorted...

also, if anyone is able to run;

```
wget -4 http://wulfy23.xyz/rpi4/builds/rpi-4_22.03.3_2.1.11.9-6_r20028_extra/rpi4.64-22.03.3-29295-2.1.11.9-6-r20028-ext4-sys.img.gz
```
and let me know what kind of speed you see... i'm pulling steady ~770KB which ain't world record breaking but is probably just workable... wondering if US/EURO/OTHER homed individuals might get better... (hostinger)

Firefishy, 2023-01-23:
> also, if anyone is able to run;
> 
> ```
> wget -4 http://wulfy23.xyz/rpi4/builds/rpi-4_22.03.3_2.1.11.9-6_r20028_extra/rpi4.64-22.03.3-29295-2.1.11.9-6-r20028-ext4-sys.img.gz
> ```
> 
> and let me know what kind of speed you see... i'm pulling around ~770KB average which ain't world record breaking but is probably just workable... wondering if US/EURO/OTHER homed individuals might get better... (hostinger)

Downloaded at 10.4 MB/s in UK.