wulfy23/rpi4 #26: fw4 errors after updating to the latest snapshot

LionHeartP, 2022-11-12:
Hi all. As per the title, something must have changed in the newest snapshot's firewall configuration. My log is full of errors;

> Sat Nov 12 11:55:11 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:11 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:11 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn6 Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:17 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:20 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:20 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn Error: No such file or directory
> Sat Nov 12 11:55:20 2022 daemon.err dnsmasq[24046]: nftset inet fw4 usrcdn6 Error: No such file or directory

Any solutions? PEBKAC perhaps? xD

wulfy23, 2022-11-12:
Thanks alot for the info and likely not PEBKAC.

We dont run fw4 so will need to look into the underlying change causing the above behavior. I'm only using ipsets for QoS on my system so its semi non critical but any peeps using 'sets' for security rules should probably downgrade or use 'release/stable'...

As a (guess at a quick) workaround removing nft style packages MAY correct the issues above but not sure how it detects firewall (ipset method) in use...

LionHeartP, 2022-11-13:
Can confirm that after updating to the firewall4 snapshot, your custom SQP script obviously doesn't work anymore, but switching to another one works. 
Also, after removing the custom ipsets (damn those looked hugely useful while accompanied by the custom sqm script ofc), no more errors!

wulfy23, 2022-11-14:
Mega sloppy play around with smudging some function across... you are welcome to try if you like (no rush or requirement... just and FYI as expressed interest);

run whole block below on oneline
```
cp /etc/init.d/firewall /etc/custom/initd_firewall.backup; curl -sSL "https://raw.github.com/wulfy23/rpi4/master/utilities/nft.user.tar.gz" > /nft.user.tar.gz && (cd / && tar -xvzf nft.user.tar.gz)
```
you'll need to restart probably firewall first, then dnsmasq then sqm if you choose to test the qos script again not that I expect it to function well... just for seeing if sets are populated I suppose... I will probably mess up your dnsmasq config again by adding sets to it so maybe back that up for easy revert if you decide to try it...

and to undo (if needed) just
```
cp /etc/custom/initd_firewall.backup /etc/init.d/firewall; sync; reboot
```

<hr>

LionHeartP, 2022-11-19:
> 

Finally gotten around to trying this.
My ipsets didn't get repopulated, but after a reboot, `user.notice SQM: ctinfo_4layercake_rpi4.qos was started on pppoe-wan successfully`

So the script is doing something.... It's definitely traffic shaping because my bufferbloat is not there, I dunno whether it's working as intended though.

Cheers!

LionHeartP, 2022-11-19:
> Mega sloppy play around with smudging some function across... you are welcome to try if you like (no rush or requirement... just an FYI as expressed interest);
> 
> run whole block below on oneline
> 
> ```
> cp /etc/init.d/firewall /etc/custom/initd_firewall.backup; curl -sSL "https://raw.github.com/wulfy23/rpi4/master/utilities/nft.user.tar.gz" > /nft.user.tar.gz; (cd / && tar -xvzf nft.user.tar.gz)
> ```
> 
> you'll need to restart probably firewall first, then dnsmasq then sqm if you choose to test the qos script again not that I expect it to function well... just for seeing if sets are populated I suppose... It will probably mess up your dnsmasq config again by adding sets to it so maybe back that up for easy revert if you decide to try it...
> 
> and to undo (if needed) just
> 
> ```
> cp /etc/custom/initd_firewall.backup /etc/init.d/firewall; sync; reboot
> ```

Finally gotten around to trying this out today. After running the command and rebooting I got:
`Sat Nov 19 17:14:32 2022 user.notice SQM: ctinfo_4layercake_rpi4.qos was started on pppoe-wan successfully`

Ipsets were not populated, but traffic shaping IS working because I don't get any bufferbloat. Not sure if working as intended though.

Cheers!

wulfy23, 2022-11-20:
thanks, dnsmasq is adding the right addresses to nftsets... so that's good, the underlying QoS/SQM is functional... but all the rules to leverage those addresses and dscp afaict may not work, or very limited bits of them got translated...

also has integrated 'blocklist' which <b>from my use of the internet this week is surprisingly speedy/snappy</b>... not sure whether this is due to dropping all the dnsmasq lists or nft... probably 65% plus is from the dnsmasq simplification if I had to guess...

so what i'd heard about nft being slower in many regards seems to be non-starter... at least on these devices... good stuff...

wulfy23, 2024-01-07:
I've renamed this topic to be a more ongoing place for anything weird or build specific regarding to 23.0x or master regarding fw4/nft.

Massive **thanks once again to @LionHeartP** for your earlier testing, without this, moving over / providing 23.0x.x builds would have been not possible or pretty rough experience...

Just a few points to summarize on the current state of alot of related stuff to the above...

a) rpi4.qos is now just a copy of 'cake.qos' I think... (i.e. I pretty much removed / disabled it but needed a 'friendly' way for everyone not to have issues so those that have it 'chosen' in sqm settings just get 'cake.qos' or whatever it is...

b) I had to add a 'dhcp' (dnsmasq) config scavenger on firstboot to trash some/all ipset entries the rpi4.qos script created, always should have put those in /tmp/dnsmasq.d not in config, but i'd written it and was tired and it was working... lesson learned (and thanks to the forum dude who pointed that out too)

can't remember if that is enabled on firstboot or everyboot, but if have custom ipset entries under the new system, and they get removed, let me know...

I will disable that crawler in a month or two or something, (give enough people time to upgrade)

c) any services / scripts etc. etc. that made use of iptables on 23.x and master(nft/fw4) are now unknown entities...

in principle, iptables commands get translated to nft commands... so they 'can/should!' work... but the heirarchy/buckets/tables which they were destined to go into in essence make all but the simplest / isolated of uses either not work or not as functional as intended...

<hr>

while standard official provides /etc/nftables.d/ (usr/share?) for custom user functionality... most people doing smaller stuff that in the past could/were advised to be done via any method (rc.local, uci-firewall include, uci-firewall_rules, other) are now being heavily advised to use just uci-firewall_rules (and ipsets))

<hr>

as with fw3/iptables builds I requisitioned /etc/firewall.user > /etc/custom/firewall.user for some build related checks and options (that not many used) except for leveraging for OS state awareness during other related step/debugging processes...

this time, for now... I've added a small HOOK to /etc/init.d/firewall that loads /etc/custom/nft.user for anything 'build specific' (setup stuff, or stuff that needs to be enabled by a user) that might be needed for general housekeeping...

I DONT/WOULD NOT USE THIS FOR ADDING 'non general streamling/optimisation' or modifying at all... (without explicit user enablement via /root/wrt.ini VARIABLE)

 and at the moment it does nothing... just there in case needed... but I call via init.d not include this time... (dunno why I did that, may change it)

<hr>

Service wise anything that is **not hugely popular** i'd say, unless it was using really bespoke (not integrated with OS naming conventions or timing etc.) rules chances are you'll see that they don't work/like they did on 22/fw3/iptables

i.e.
upnp (was alot of work but dunno about it), transmission, docker, nft-qos?, etc. etc. etc.

most popular services from what i've seen from some superficial scans seem to be working really well (banip, pbr, etc. etc.)

<hr>

right now I'm not planning on doing much, but a routers FW will inevitably, enter the realm in terms of operation and needs... 
(but customisation / efficiency / bugfixing are always of interest) 

anyway, don't really expect anyone to read all that jibberish but was kinda necessary to put down a 'state of affairs' and 'call to action' regarding issues/needs moving forward with the iptables > nft migration

<hr>

i.e. just a lil example of running firewall restart manually on my loooooong held / munged fw config results in carps of (benign output)... related to the change...
(i should really start fresh ;)

-disabled sections ok
-old includes no longer running [didnt need/use them but you may have one / some]
**(requires 'option fw4_compatible 1' ~ after you hand test!)**
-new nft includes ok-ish need to peruse
-reload not supported

```
[root@rpi-dca6325631 ../_argon_2.3.1 59°] /etc/init.d/firewall restart
Section @rule[1] (BLOCKDoT4) is disabled, ignoring section
Section @rule[2] (BLOCKDoT6) is disabled, ignoring section
Section @rule[7] (mqqtoggle-test1) is disabled, ignoring section
Section @rule[8] (mqqtoggle-fish-extra) is disabled, ignoring section
Section @rule[14] (Support-UDP-Traceroute) is disabled, ignoring section
Section @include[0] option 'reload' is not supported by fw4
Section @include[0] is not marked as compatible with fw4, ignoring section
Section @include[0] requires 'option fw4_compatible 1' to be considered compatible
Section agh option 'reload' is not supported by fw4
Section agh is disabled, ignoring section
Automatically including '/usr/share/nftables.d/table-post/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_forward/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_input/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_output/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_postrouting/30-pbr.nft'
Automatically including '/usr/share/nftables.d/chain-post/mangle_prerouting/30-pbr.nft'
```

FIN

LionHeartP, 2024-01-07:
Thank you for the update!
While the update finished successfully (?) there seems to be no DNS resolving happening? Not sure what's wrong, but I can ping IPs successfully while no pages open on any device in the network.

Can't see anything suspicious in the logs. What could have triggered this behavior?

wulfy23, 2024-01-08:
Thanks for the report... and bummer network did not come up.

Most likely thing would be a service I don't use (https-dns?), or maybe something stale in the dnsmasq config...

from memory when my masq didn't come up last time (similar underlying system), that was when I added some code to tear down stuff other services put in there (but https dns is the only one that has prevented functioning clients)... on upgrade and firstboot... chance those things I added messed it up too...

(note: a dmesg and a logread from the effected system might have shown some obvious signs but yeah, given timing / readership etc. starting fresh is probably better anyways)

anyway, glad you started fresh and are back up and running... were I not the maintainer, I should be doing the same!