wulfy23/rpi4 #31: 2.1.11.9-6 - argon theme not working?

molly78, 2023-01-23:
It’s stuck on bootstrap, argon doesn’t seem to wanna load

thanks wulfy23!

wulfy23, 2023-01-23:
yeah, there had been some possibility of complications so simplest thing for me to do was remove it temporarily as I was in a rush to get it out.
(and when it failed for me last time on master I got nothing... no webUI at ALL, which suspect was not all argons issue if at all... something between the ucode changes / package requirements and my scripts likely also interacted badly)

suspect it would have / will work (for 22.03), and planned/plan to test it out properly this week and push and updated build sometime likely this weekend

feel free to try install it manually before then tho', think there is some quick install commands somewhere on the forum.

molly78, 2023-01-24:
Ok no problem wulfy23. It’s not a biggie so I can wait :-)

wulfy23, 2023-01-25:
so tested it and uploaded 2.1.11.9-12 with the legacy/typical build argon (without config app) included...

just posting here mostly to highlight its 'typical current behavior'... which i'm pretty sure is no change from 22.03.2 ( afaik the updatecheck bar last was visible in argon 22.02.1)

1. Updatecheck bar is not shown ( but the luci > system > commands simple workaround / alternative should now be available )
2. On first (pre)login after upgrade, the lefthand menu is visible rather than the expected 'bing picture' splash page (after a reboot picture splash should then work / works for me)

wulfy23, 2023-05-23:
ok, good-ish news, current (future) master build(s) r23001 ( 7.1.93-12 ) have been able to get argon back and bumped to current version 2.3.1

it's vanilla tho' so upgrades / checking would have to be via **system > custom commands**... (aka no top bar) and it's a bit 'bubblier' out of the box... (or however you prefer)

point 2 in above post seems mostly resolved also, seems to work from first install although I have needed to refresh after install as css looks like does not load. not sure if others get this... ( it's like once per install or session or something, once you refresh it does not come back )

will close this issue as argon is back in, and i'll keep testing on newer builds but feel free to open new one if more info comes to light or non-exact issues arise.

thanks for the report