wulfy23/rpi4 #34: Packet Steering unticked after a couple visit

norbertjoni, 2023-02-06:
In the network - interface - global network option if I tick the Packet Steering after a couple of GUI visits it's off. Not sure why and I don't see anything in the log. 

Has anyone got a similar issue? 

irqbalance is enabled but if I disable the service I get the same issue.

anipad, 2023-02-06:
Same here, but it correspondes with my VPN interface resetting => no packet steering until I re-enable it.

norbertjoni, 2023-02-12:
I have a VPN too, but in the original fw was not an issue.

neilo1, 2023-02-20:
Yes, it is disabled after every reboot or power failure. 

[image]

wulfy23, 2023-03-09:
hey guys, thanks for the feedback, most things should leave a trail in `dmesg` ... but it's spammy so unless you see something that really stands out it may not be worth posting here... just yet anyway....

I should have a little time this weekend to have a quick poke against the 'new' stable as these things otoh seem to originate around there... 

I will probably 'push' some part of my tree (i.e. most obvious code I have that might be interacting with this) such as this ( i.e. bits that I would not mind, or were very useful to others in the past )  in the unlikely event someone has lower level fault finding skills to assist or fix... but honestly it's probably due for a total rewrite...

I'm finding I have less and less free time of late... and for you guys it might be time to jump ship. ( But I DO APPRECIATE/STILL NEED THE REPORT!!! so THANKS! ), most time I do have I'd like to focus on gutting stuff from 'master'/head/whatever and or cleaning up the really core parts of the build... so alot of these 'helper' elements ( read: where several use cases increase the complexity of application ) i'll probably just farm out / open up / seek third party sources and ensure there is a way to disable them rather than fix (at least that would be the initial approach.

nikohd12, 2023-05-17:
I am getting the same error from a fresh install. Used to be unticked on an old release and just upgrading. But one day I decided to clean flash to remove some errors and do some cleaning on my free time.

nikohd12, 2023-05-17:
I have WireGuard enabled if it helps.

wulfy23, 2023-05-25:
so will attempt to 'fix'/improve this on the weekend...

long story short, the legacy reasoning behind that (assuming it was coded in the right place/logic) was that if the built in perftweak script is managing your steering params then it will turn off that UCI setting.

(AKA you probably still have steering, just my preferences on that)
i.e. I do not have it ticked but
```
#dmesg | grep perftweaks
[  644.693692] rpi-perftweaks.sh[bg]> ETH0INT:  36:f>1 37:f>2
[  644.749975] rpi-perftweaks.sh[bg]> LANIFACE:eth0 WANIFACE:eth775c13
[  644.797554] rpi-perftweaks.sh[bg]> sysval: eth0/queues/tx-0/xps_cpus[0>1]
[  644.845342] rpi-perftweaks.sh[bg]> sysval: eth0/queues/tx-1/xps_cpus[0>2]
[  644.893396] rpi-perftweaks.sh[bg]> sysval: eth0/queues/tx-2/xps_cpus[0>4]
[  644.940944] rpi-perftweaks.sh[bg]> sysval: eth0/queues/tx-3/xps_cpus[0>4]
[  644.988448] rpi-perftweaks.sh[bg]> sysval: eth0/queues/tx-4/xps_cpus[0>2]
[  645.035909] rpi-perftweaks.sh[bg]> sysval: eth0/queues/rx-0/rps_cpus[0>7]
[  645.083292] rpi-perftweaks.sh[bg]> sysval: eth775c13/queues/rx-0/rps_cpus[0>7]
```
tailored packets steering is working...

but yeah, the current logic is probably deficient... and especially as you'd have to dig a bit to find or even understand how/when 'my settings' are applied the below seems like a good way forward

will try minimum;

a) to not tick or untick that
b) if it is ticked, I'll apply my custom ones afterwards (the stock steering values), make sure it's logged, and make sure there is a way to disable my stuff...

seems like a fairly clear plan...

wulfy23, 2023-05-27:
ok part a is done, as updatecheck bar does not run all the time now... you'll have to go to **luci > system > custom commands > UPDATECHECK [run]**

to apply hotfix

nikohd12, 2023-05-27:
Hi @wulfy23,

That was quick, thank you so much. I have encountered this problem when running UPDATECHECK

[image]

nikohd12, 2023-05-27:
Command worked for the alternative.

[image]

Will test for 2 days.

wulfy23, 2024-01-07:
I think this is semi-sorted or best to start new if other stuff crops up, but re-open if needed

closed 2024-01-07 by wulfy23