wulfy23/rpi4 #35: WAN ip renews and need to reboot

neilo1, 2023-02-11:
When wan IP address renews on its own then the internet stops working and I have to reboot the router for it to work properly. I think I have been facing this issue for the last 2 or 3 builds.

wulfy23, 2023-05-25:
hi mate,

apologies I did not answer this quicker... basically it's related to the simple adblock package and how it leaves entries in dnsmasq.conf / starts up.

short story is I tried to get the maintainer to address the former ages ago.

basically ditch the package or find the line with
```
/var/run/simple-adblock/dnsmasq.servers
```
in /etc/config/dhcp
and remove it... restart the service dnsmasq... restart simple-adblock etc. etc. until masq starts properly...

there was also a bug with that path not playing ball with procd-ujail but again, that would be a maintainer issue... i did work around it for several months... just became to much hassle...

neilo1, 2023-06-04:
Thank you very much. 😀

wulfy23, 2023-06-05:
hi neilo1, i'm guessing that the thanks is because the latest build ( 22.03.5 update 1 ) maybe improved your connection issues? ( masq not starting possibly because of simple-adblock or it starting and simple can't bootstrap or something like that with possible pppoe delays thrown in )

if your just thanking for the build no problemo...

if your stuff has improved, I did make some changes related to your issue...

basically prior to upgrade ( would not have ran for you this time ) and early on firstboot i've attempted to flush simple-adblock / https-dns "stale" entries from the dnsmasq/dhcp config...

one semi-downside of note is that during early boot ( first 5mins ) the router itself may use "normal" dns services... but the way the build is setup for many different people it's the a compromise I kinda have to make to ensure networking is a little more robust / predictable during upgrade...

anyway, thanks for the message... while I can't rule out issues arising from changes above, if it helps you or anyone else then it would have been worth it...

neilo1, 2023-06-05:
For now, I have disabled simple-adblock and banIP. I was unable to use google meet and some other websites.

wulfy23, 2024-01-03:
I think the package is no more? and / or 23.x + time sort of place this report in the stale / hard to correlate with much moving forward so i'll close...

but open another/this issue if it comes back or you get more symptoms... thanks neilo...

closed 2024-01-03 by wulfy23