wulfy23/rpi4 #38: release bump / 22.03.4

moodyblue, 2023-04-14:
Hi Wulfy, would it be possible to update your build to 22.03.4 ?

Thanks a lot

molly78, 2023-04-15:
wulfy23, you’re our only hope!

Synapse07, 2023-04-27:
2nd this.

wulfy23, 2023-05-07:
thanks for the info. just uploaded a quick and dirty build ( 2.1.15-2 ), no changes on my end afair...

only had time so far for a quick install test so fingers crossed not too much degradation in other areas

wulfy23, 2023-05-07:
what the... lol

[https://github.com/wulfy23/rpi4/raw/master/misc/whatthe22035statswifi.png](url)

Synapse07, 2023-05-07:
just wanted to say thanks.

Synapse07, 2023-05-07:
no banip?

wulfy23, 2023-05-07:
thankyou! 

legacy ban packages **uploaded to this repo under /utilities for manual install.**.. 2.1.15-5 added them back...

just on this... those packages i've been dropping in the last ipt compatible versions for maybe most of 22.03 (5ish build versions)...  as such, the lists they pull and whatnot may start/already see 404's and whatnot...

happy to mod the package if anyone knows/want's blocklist url change or similar for 22.03.6 etc. (or anyone else using fw3)

thanks again, to creator of that package for it. i'm mindfull that using/distributing static/dated versions of others code might not be polite. but hey, if it's working then it also extends the value of their efforts.

Synapse07, 2023-05-07:
the addition of the latest blocklists would be very nice. thanks for everything.

molly78, 2023-05-07:
> grandson born today. woohooo!

Wow congrats!!!

wulfy23, 2023-05-15:
ok so i've setup an (ipt/22.03.x for now) banip blocklist called 'wulfyblock'... should show up the next time you login to luci if anyone wants to try...

(at the moment it's a de-duplicated 'everything from the new banip' but might throw some additional stuff in there / or make another list or two from other sources)

I've seen list servers get (understandably) restrictive in the past (and this time I tested a few lists) so figure may as well, take some of the burden off / provide some redundancy for them...

if anyone has some firehol etc. lists that seem to differ from standard openwrt ones... will take a look and merge / incorporate where feasible...

on the same topic I run some custom params in **/root/wrt.ini** that relate to the above somewhat...

```
DLWRAPPER=1
DLWRAPPER_CHECKPERIOD="hourly"
```
these settings are 'applied' during firstboot but you could also just run
```
/etc/custom/firstboot/051-banipng
```
to activate them without having to upgrade...

in it's current form... it's basically just a 'blocklist cache' that helps a little with bandwidth / redundancy / reload speed for large lists...

wulfy23, 2023-05-15:
k so i've setup an (ipt/22.03.x for now) banip blocklist called 'wulfyblock'... should show up the next time you login to luci if anyone wants to try...

(at the moment it's a de-duplicated 'everything from the new banip' but might throw some additional stuff in there / or make another list or two from other sources)

I've seen list servers get (understandably) restrictive in the past (and this time I tested a few lists) so figure may as well, take some of the burden off / provide some redundancy for them...

if anyone has some firehol etc. lists that seem to differ from standard openwrt ones... will take a look and merge / incorporate where feasible...

on the same topic I run some custom params in /root/wrt.ini that relate to the above somewhat...

```
DLWRAPPER=1
DLWRAPPER_CHECKPERIOD="hourly"
```
these settings are 'applied' during firstboot but you could also just run

/etc/custom/firstboot/051-banipng
to activate them without having to upgrade...

in it's current form... it's basically just a 'blocklist cache' that helps a little with bandwidth / redundancy / reload speed for large lists...

```
[root@rpi-dca6325631 /usbstick/TXT/twrp 50°] /etc/init.d/banip status
::: banIP runtime information
  + status          : enabled
  + version         : 0.7.10
  + ipset_info      : 14 IPSets with 450960 IPs/Prefixes
  + active_sources  : bogon, debl, doh, drop, dshield, edrop, whitelist, wulfyblock
  + active_devs     : eth775c13
  + active_ifaces   : wan, wan6
  + active_logterms : dropbear, sshd, luci, nginx
  + active_subnets  : 111.x.x.x/22, 2111:xxx:ffff:12:ccc:uuu:5555:7777/128
  + run_infos       : settype: src+dst, backup_dir: /restorefiles/service-dpata/banip/backup, report_dir: /tmp/banIP-Report
  + run_flags       : protocols (4/6): ✔/✔, log (src/dst): ✘/✘, monitor: ✘, mail: ✘, whitelist only: ✘
  + last_run        : restart, 1m 0s, 1902/1349/1396, 15.05.2023 20:48:56
  + system          : Raspberry Pi 4 Model B Rev 1.1, OpenWrt 22.03.5 r20134-5f15225c1e

[root@rpi-dca6325631 /usbstick/TXT/twrp 50°] free
              total        used        free      shared  buff/cache   available
Mem:        1902148      181092     1392484       77384      328572     1439692
Swap:             0           0           0
```

wulfy23, 2024-01-03:
regarding 22 ( .05.6 ) I kinda bit the bullet and hopped over to nft ( 23.05 )...

this will make life a gazillion times easier for me and 22 is only supported till april so...

BUT... If anyone absolutely NEEDS iptables ( 22.x.6 )... and hasn't yet upgraded to 23.05.2 (or want's to downgrade and hand edit / restore configs or install factory and restore ) then I will build an image...

wulfy23, 2024-04-06:
closing, if anyone wants iptables re-open... many packages have changed / switched to nft only logic... so running iptables is more of an individual thing... might be able to build it... but def, cannot 'support' it across many peoples setups...

closed 2024-04-06 by wulfy23