wulfy23/rpi4 #40: IPV6 Disabled on network devices with wan6 configured

ctaylor4874, 2023-08-08:
Hello,

First off, thank you for all your work on this. I seriously appreciate it!

I'm having an issue where I have `wan6` interface configured and IPV6 working on my local network, but in the device configuration within Luci, the `Enable IPV6` option is set to `Automatic (disabled)`. I have a Nanopi R4S as well, and with the same interface/dhcp configuration shows `Enable IPV6: Automatic (enabled)` for the wan and br-lan devices. It _could_ be a bug on the UI side, but I'm skeptical. Are there any configuration changes I need to make in OpenWRT for it to detect IPV6 capability, or should I just ignore this?

Beyond that, any additional tips for IPV6 configuration?

### RPI4 / Wulfy build
[image]
[image]

#### `cat /etc/config/network /etc/config/dhcp`
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ipaddr '10.0.0.1'
	option delegate '0'
	option ip6assign '64'

config interface 'wan'
	option proto 'dhcp'
	option device 'eth1'
	option peerdns '0'
	list dns '1.1.1.1'
	list dns '1.0.0.1'

config device
	option name 'eth1'

config device
	option name 'eth0'

config interface 'wan6'
	option proto 'dhcpv6'
	option device 'eth1'
	option reqaddress 'try'
	option reqprefix 'auto'
	option peerdns '0'
	list dns '2606:4700:4700::1111'
	list dns '2606:4700:4700::1001'

config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
	option dnsforwardmax '2300'
	option min_cache_ttl '270'
	list address '/router/10.0.0.1'
	option cachesize '1000'
	option rebind_protection '0'
	option port '54'
	option noresolv '0'
	list server '10.0.0.1'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	option leasetime '24h'
	list domain 'home'
	list dhcp_option '6,10.0.0.1'
	list dhcp_option '3,10.0.0.1'
	list dns '***********'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
```

### NanoPi R4S (Note: WAN device is `eth0` on this device)
[image]
[image]

#### Configs
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth1'

config device
	option name 'eth1'
	option macaddr '**********'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option delegate '0'
	option ip6assign '64'
	option ipaddr '10.0.0.1'

config device
	option name 'eth0'
	option macaddr '*******'

config interface 'wan'
	option device 'eth0'
	option proto 'dhcp'
	option peerdns '0'
	list dns '1.1.1.1'
	list dns '1.0.0.1'

config interface 'wan6'
	option device 'eth0'
	option proto 'dhcpv6'
	option peerdns '0'
	list dns '2606:4700:4700::1111'
	list dns '2606:4700:4700::1001'

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

config device
	option type 'bridge'
	option name 'docker0'

config dnsmasq
	option localise_queries '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
	option noresolv '0'
	option cachesize '1000'
	option rebind_protection '0'
	option port '54'
	list server '10.0.0.1'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	option leasetime '24h'
	list dhcp_option '6,10.0.0.1'
	list dhcp_option '3,10.0.0.1'
	list dns '************'
	list domain 'home'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
```

wulfy23, 2024-01-03:
Thanks for the query...

Yeah, doesn't look like anything major (build related / obvious problems)...

Mine shows the same thing and works...

However, your config declares device br-lan with eth0 port correctly, then you have a second ( superfluous? ) device named eth0 which is redundant / uneeded / conflicts... If I had to guess as you only declare a name in those devices, they don't seem to be causing any issues here if I had to guess...

i.e.

normal
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
```

incomplete / un-needed / other entries
```
config device
	option name 'eth1'

config device
	option name 'eth0'
```

in other words, i'd probably need more info (if there is an actual functional deficiency) but it looks fine (subject to standard config nonclemature) to me...

wulfy23, 2024-01-07:
please re-open this if you come back and need to

closed 2024-01-07 by wulfy23