wulfy23/rpi4 #45: Migrating to standard release?

grischard, 2023-12-15:
Our friend wulfy23 seems to be busy with other things. With no release coming for 23.05 and updates stopping, how many of you are switching to the mainline releases? What's the easiest way of doing this? Can you upgrade the latest wulfy23 release to 23.05 without breaking everything?

neilo1, 2023-12-21:
No Idea.

tedsears, 2024-01-02:
According to this thread it is doable, though I have not yet tried it.

https://www.reddit.com/r/openwrt/comments/uggcse/transition_from_wulfy23_build_to_official_build/

While I have very much enjoyed these builds I am starting to worry about missing out on newer fixes etc.

wulfy23, 2024-01-03:
Just a note on this guys, **with 23.05 + master (nft post)**, particularly because most are using adblock / banip / or whatnot, the current build is kind of a "breakpoint" when it comes to configs.

i.e. alot of stuff is getting stale so your /etc/config/network will work across migration but other than that, most service files should probably be started fresh as a lot of services have changed / don't exist etc. etc.

if you are going from 22.03.x to 22.03.6?+ and never installed 23.05 then previous advice should be the same... 

the key thing to always remember when migrating from my builds is to NEVER use a standard openwrt backup restore (i need to expose the old custom backup logic from the updatecheck bar so you can export a 'friendly' importable backup, will make a note of this) as it contains /boot/cmdline.txt which will prevent your system from booting unless you put the sdcard into a computer and adjust the root=/dev/mmcblk0p2

neilo1, 2024-01-05:
> Just a note on this guys, **with 23.05 + master (nft post)**, particularly because most are using adblock / banip / or whatnot, the current build is kind of a "breakpoint" when it comes to configs. [...]

How do I update to a new fix when the upgrade bar is gone?

neilo1, 2024-01-05:
> Just a note on this guys, **with 23.05 + master (nft post)**, particularly because most are using adblock / banip / or whatnot, the current build is kind of a "breakpoint" when it comes to configs. [...]

How do I upgrade to a new fix when upgrade bar is gone?

wulfy23, 2024-01-06:
Hi Neilo,

luci > system > commands

**To check if update is available**
click RUN under UPDATECHECK

To update
**click RUN under UPDATE**

(sadly the step above will just seem like it's taking forever / hanging but you kinda need to trust it's doing it's thing unless you wanna run it from a command line)

wulfy23, 2024-01-07:
For safe restore to clean standard (not my builds) a (hopefully safe/restorable to stock) backup can be generated similar to this (paste) via ssh;

```
tar -cvzf /www/backup.tar.gz $(sysupgrade -l -q | grep -Ev '(/etc/passwd|/etc/shadow|/etc/group|/etc/config/rpcd|/etc/config/uhttpd|/root/tasks|/root/backup|/restorefiles|/etc/custom|/etc/tvheadend|/etc/php8|/etc/firewall.user|/boot/cmdline.txt|/etc/profile.d|/etc/sysupgrade.conf|/etc/rc.local|/etc/simple-adblock.servers.gz|/etc/profile|/etc/shells|/etc/shinit)') 1>/dev/null 2>/dev/null 
echo "Download backup from http://router/backup.tar.gz / https://router/backup.tar.gz #10mins only"
(sleep 300; rm /www/backup.tar.gz 1>/dev/null 2>/dev/null)&
```

then download it from your browser...

if you want to try another trick that may/may not work, rather than starting with a stock factory...

you *MAY* be able to do;
```
cp /www/backup.tar.gz /boot/sysupgrade.tgz
sysupgrade -n /tmp/standard-SYSUPGRADE.img.gz
```

or similarly, you could **flash the factory then put into a PC and copy backup.tar.gz to partition 1 (boot) as sysupgrade.tgz)** this would probably work as opposed to the method above...

if you are totally unable to use ssh, next build will have this in luci > system > commands > 'rpi-support.sh safebackup' for download with browser

<hr>

if anyone tries this, would be good to hear how it goes...

tedsears, 2024-01-07:
wulfy23, thank very much for the ongoing followup!

I have generated a backup using your tar command on the commandline, and will keep that around.   Basically I should be able to to install a vanilla openwrt build, and then just restore this backup.tar, and at least have all the basic setup?  It looks like most of that stuff is just stored in /etc anyway.    I just would hate to have to recreate all my vlans, static ips, wireguard connections, etc, etc, etc.    

Overall I really like your builds, and since you have a new one up, I may just stick with you for a bit longer.   But it's nice to have an "off ramp" if needed.

I am currently on your build of OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598.   It sounds like from your above responses to Neilo1 that it should be safe to do the upgrade to the "release [newer-major] 3.0.6-25(2.1.11.9-12)" that is returned by updatecheck, although i will need to start over with adblock and some of the other service configs?   Mostly I use dynamic dns, DNS HTTPS Proxy Settings, and adblock.   I suspect setting those back up shouldn't be too terrible.

Thanks again for the builds, and your willingness to answer questions!

Ted

wulfy23, 2024-01-08:
g'day,

yeah your summary is pretty on point. just upgrade and keep an eye on the log to spot anything you run that's acting weird.

https-dns i've not tested so that's up in the air... (from memory they made some tweaks to some of the default config... added that dns instance to choose from I think? unless that was always there)

unfortunately the only way to get the new config file is to re-install the package, hmmmm... a command line is gonna be necessary at some point as either **--force-maintainer** on re-install

or

cp /etc/config/PACKAGE-opkg /etc/config/PACKAGE

will be needed... ( or just move / deletion of your old config before re-install )

might write a luci-app-command this weekend that dumps 'vanilla oem configs' for popular services by name this weekend...(or runs opkg --force-maintainer) tough predicament... (want to upgrade and keep network and nothing else... config wise even then i'm probably making too many decisions for the user)