wulfy23/rpi4 #47: Wifi not starting automatically starting on boot

Synapse07, 2024-01-14:
title.
I have to click restart from luci. im on the latest build.

wulfy23, 2024-01-15:
thankyou...

will investigate a little more this weekend / ponder how to best handle... in the meantime... this may resolve for now...

```

opkg update
opkg install --force-reinstall cypress-firmware-43455-sdio
opkg install --force-reinstall brcmfmac-nvram-43455-sdio

pushd /lib/firmware/brcm
ln -s /lib/firmware/cypress/cyfmac43455-sdio.bin brcmfmac43455-sdio.raspberrypi,4-model-b.bin
ln -s /lib/firmware/cypress/cyfmac43455-sdio.clm_blob brcmfmac43455-sdio.raspberrypi,4-model-b.clm_blob
popd
```

Synapse07, 2024-01-15:
for pushd i get
`[root@RPi4B /]# pushd /lib/firmware/brcm
-ash: pushd: not found`

same with popd

wulfy23, 2024-01-15:
```
cd
```
will do instead (fixed)

Synapse07, 2024-01-15:
Thank you bossman.

closed 2024-04-07 by wulfy23

Homas, 2024-04-17:
Just deployed `rpi-4_23.05.3_3.1.5-5_r23809_extra` build.
WiFi interface wasn't available. I had to reinstall the packages (the symlinks were there) to make it working
```
opkg update
opkg install --force-reinstall cypress-firmware-43455-sdio
opkg install --force-reinstall brcmfmac-nvram-43455-sdio
```