wulfy23/rpi4 #49: In latest build, Pi400 wireless is not working. Wireless tab not showing up in Luci.

aki-mizu, 2024-01-22:
@wulfy23 
According to this release note in .25:
`rm local brcmfmac bins -> may effect setup for 400/CM4 wifi depending on what
                    UPSTREAM now includes in image... can putback if anyone has issues `
As I tested in Pi400, wireless does not work anymore due to lack of brcmfmac firmware, it was working so far until this release. Could you please put them back for Pi400/CM users? Thanks!

wulfy23, 2024-01-24:
thanks, yes it appeared that 400 and CM4 would get hit by the purge of the old firmware files we were carrying...

can you be more specific about 'them'? as I poked around a little and it seems pifoundation/limited uses 43456? for 400 and appears like owrt links / uses standard 43455... ? is there a specific distro / firmware blob you prefer?

owrt does use newer blobs tho', so yeah, spent 2 hours looking around at what distros are supplying and alas its not so simple as drop and go... ( pifoundation/limited links about 4 places together )...

On the plus side, I have a 400 I can test with... if you don't specify will just push the standard (actually i'll push it now and you can try it out if you feel like it with)

```
sysupgrade -R <img>
```

aki-mizu, 2024-01-24:
Yes, both CM and 400 uses 43456 as it's indicated here:
https://github.com/RPi-Distro/firmware-nonfree/tree/3db4164cfd89e6d9afb7ebc87607b792651512df/debian/config/brcm80211/brcm
And also here:
https://github.com/geerlingguy/pi-router/issues/4
So we need these 6 files maybe for supporting both devices.
brcmfmac43456-sdio.raspberrypi,4-compute-module.bin

brcmfmac43456-sdio.raspberrypi,4-compute-module.clm_blob

brcmfmac43456-sdio.raspberrypi,4-compute-module.txt

brcmfmac43456-sdio.raspberrypi,400.bin

brcmfmac43456-sdio.raspberrypi,400.clm_blob

brcmfmac43456-sdio.raspberrypi,400.txt

wulfy23, 2024-04-06:
added into 23.05.3(5+)

closed 2024-04-14 by wulfy23