wulfy23/rpi4 #50: Realtime Graphs - Realtime Rate page shows no data

markjsones, 2024-01-29:
I'm running OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-24.006.68745-9128656
It's my first OpenWrt box, on a Pi4.

All the important stuff works great, and I have learnt a lot.  nlbwmon is cool, and working.
Most of the graphs under Status->Realtime Graphs work fine, but "Rate" tab does not.

It just says 'Collecting data...'
Visiting http://192.168.1.2/cgi-bin/luci/admin/status/realtime/rate_status, I get
{ "nftables": { } }

This suggests the UI is working fine, but underneath something isn't able to collect any data.

I've searched around in the bowels of the thing for hours, but can't find anything to reconfigure or test to help me get any traction.
I've searched the internet, and again failed to find anything that applies in this circumstance.

Any ideas?  Thanks

markjsones, 2024-01-29:
Should say, Load, Traffic and Connections tabs all work in stats.
I'm not using any wireless devices, so that tab is uninteresting.

molly78, 2024-02-08:
Rate requires openwrt to be your dhcp server

you can manually add the ip addresses as Wulfy23’s posted about it once. Now where is that

markjsones, 2024-02-09:
Thanks, that's interesting.  I'm doing DHCP and DNS elsewhere.  On the Openwrt, I've already added MAC addresses, names and IPs into /etc/config/dhcp, so that the names and IPs show up in the nlbwmon dashboards.  But it's not helped on the realtime - rate page.

In /etc/config/dhcp, they are in this format:-

config host
	option name 'anamexxxx'
	option mac 'xx:xx:xx:xx:xx:xx'
	option ip '192.168.1.xxx'

Been looking for the post you mention but can't see it.

markjsones, 2024-02-12:
Did you find that post @molly78 ?
I had a search and couldn't see it.

molly78, 2024-02-14:
> Did you find that post @molly78 ? I had a search and couldn't see it.

Man alive, no. It could be buried in the main GitHub super long post maybe?

markjsones, 2024-02-15:
I'm totally stuck. I can't find the post that @molly78  speaks of, and there's nothing on the internet that helps.

Is @wulfy23  ever about on here?

molly78, 2024-02-16:
> I'm totally stuck. I can't find the post that @molly78 speaks of, and there's nothing on the internet that helps.
> 
> Is @wulfy23 ever about on here?

Yeah sorry there. It was a side quest in a post I came across but can’t seem to find it again. Gah

where are you @wulfy23. You’re our only hope!

wulfy23, 2024-04-06:
I think this is probably the link / info molly78 so kindly mentioned.

https://forum.openwrt.org/t/rpi4-community-build/69998/1560

markjsones, 2024-04-06:
Hi, I've used this method to add all my IP addresses into the config.  THANKS.
So I can now see them listed in the UI 'rate' window.
However, there are no counts/rate information being updated.
I will dig into the services to make sure it's all running.

markjsones, 2024-04-06:
I'm running default so br-lan for 'LAN' zone (comprised of eth0).  WAN side is eth1, well eth.911 more accurately.
I have the 'enable' config etc, but cannot see any unix processes running.  What processes should I expect to be running at this point to underpin this?  ps |grep nft and ps |grep qos show nothing relevant

wulfy23, 2024-04-06:
basically the "service" has two components...

-on startup it sets up some nft "containers" for each function
-hotplug insertion of hosts into those containers in the case of the "monitor" function

(aka, there is no background process that runs)

in your case, the second function is manual, so in the output of ```nft list ruleset``` you'd just see counters being updated in the 'nft-qos-monitor' tables (upload and or download) i.e.

```
table inet nft-qos-monitor {
	chain upload {
		type filter hook prerouting priority filter; policy accept;
		ip daddr 10.2.3.44 counter packets 0 bytes 0
		ip daddr 10.2.3.45 counter packets 0 bytes 0
		ip daddr 10.2.3.122 counter packets 674209 bytes 986466221

```

markjsones, 2024-04-07:
OK, odd one this, but I think it's sorted.  I think the problem was related to the 'type filter hook....' row, which looked different in the configuration on my openwrt instance.

I've deleted the table with 'nft delete table inet nft-qos-montior', then restarted the services (which recreates the table).
I then populated it again with all my IPs, and it seems to be working!

Thanks Wulfy23!

wulfy23, 2024-04-08:
interesting, thanks for the info will look at that if I ever get faster speeds or have a system setup for such a test internally. otoh, it should not cause such a hit...

markjsones, 2024-04-08:
OK cool.  FYI, my setup is quite device heavy, so I've added the best part of 500 IP addresses across 192.168.0.1/23 (ie 192.168.0.x and 192.168.1.x), which is probably not typical.  I suspect performance would be dramatically better with even only half the number of IP addresses in the tables.

closed 2024-04-09 by markjsones