wulfy23/rpi4 #52: Synology rt2600ac source code

webysther, 2024-04-11:
Hi, I can't find in the commit history the code provided to rt2600ac, any chance to send to me or provide with a full disclore about risk? I want to keep the work on a rt2600ac I have in the next few months.

wulfy23, 2024-04-14:
work I did on this device was a long time ago and non-upstreamable due to usb nature of it.

all you really need is an nbg dts to get started... however, i've re-pushed what I did at the time, which is **mostly unusable** in it's current form... although, the initramfs + uboot commands may assist you / others in getting up an running on this device.

good luck with it! ( p.s. my work does not have FUNCTIONAL upgrade / install to mmc featureset so BE VERY CAREFUL and any attempt to do so is at your own risk. )

closed 2024-04-14 by wulfy23

webysther, 2024-04-22:
Thanks man! Just sometimes the start point is all we need, thank you for the previus effort!